tuanpmt/esp_mqtt #120: Disconnect after publish

harryd100, 2016-09-10:
Hi, I converted one of my existing working (SDK 1.5.2) project to SDK 2.0.0 using the latest esp_mqtt, but see some strange behaviour. After publishing a message, every 10 sec, the MQTT connection is disconnected. Wifi stays up as I can still ping the ESP. Serial log says:
MQTT: Published
TCP: data received 5 bytes
TCP: data received 123 bytes
TCP: Disconnected callback
MQTT: Disconnected
TCP: Free memory
TCP: Connect to ip  192.168.24.250:1883
TCP: Reconnect to: 192.168.24.250:1883
MQTT: Connected to broker 192.168.24.250:1883
MQTT: Sending, type: 1, id: 0000
TCP: Sent
TCP: data received 4 bytes
MQTT: Connected to 192.168.24.250:1883
MQTT: queue subscribe, topic"/Device/ESP_0432AC/in", id: 1
MQTT: queue subscribe, topic"/Device/ESP_0432AC/out", id: 2
MQTT: Sending, type: 8, id: 0001
TCP: Sent
TCP: data received 5 bytes
MQTT: Subscribe successful
MQTT: Sending, type: 8, id: 0002
TCP: Sent
TCP: data received 41 bytes
MQTT: queuing publish, length: 123, queue size(0/2048)
MQTT: Sending, type: 3, id: 0000
TCP: Sent
MQTT: Published
TCP: data received 5 bytes
TCP: data received 123 bytes
TCP: Disconnected callback
MQTT: Disconnected

and so on

Any idea what is happening here?

harryd100, 2016-09-10:
never mind, found the issue

closed 2016-09-10 by harryd100

fargiolas, 2017-07-07:
> never mind, found the issue

Hi I've been seeing a similar issue, it would be really helpful if you could add a comment about what you did to fix it...