tuanpmt/esp_mqtt #123: Queue on Flash memory

asadislam94, 2016-10-24:
Is it possible to create the queue on flash memory to make sure we still have data during power failure?

tuanpmt, 2016-10-25:
Maybe we can use a large portion of flash to do this by writing circle small data.

someburner, 2016-10-25:
IMO this wouldn't really help anything. The only reason your buffer should be full is if WiFi is disconnected, not a power failure. Since I assume you can't predict when the failure occurs, it could happen at any random point, like during the middle of a flash read/write operation and then you would potentially be sending corrupt data. Unless you mean a power failure that affects your AP but not your ESP?

asadislam94, 2016-10-25:
There can be a situation where a power failure occurs to ESP during a network outage. If this instance occurs then you may want to keep the important publish data saved for when the device turns back on.

However I just realized that this doesn't need to be implemented on the queue. We can just keep the "important data" in flash and send it using QoS=2. If the data is sent correctly it can be deleted otherwise on boot-up we can have a check to see any data on flash and send it.

closed 2016-10-26 by asadislam94