tuanpmt/esp_mqtt #129: lost messages with QoS == 2

st0ff3r, 2016-11-28:
From debugging the transmission from espmqtt to mosquitto broker it seems that if the the first message with type MQTT_MSG_TYPE_PUBLISH got lost and never received by the broker there is no retransmission and the message is lost. Is that right understood?

Second question: in mqtt_msg_init() when called from mqtt_tcpclient_connect_cb() shouldn't connection->message_id be saved if not clean session and QoS > 0, so message id is not going to overlap with messages from the session on the broker?

tuanpmt, 2016-11-28:
i'm porting this library to Arduino: https://github.com/esp8266/Arduino/pull/2715
with outbox and have tested with qos=2.
About qos, you can check here for your understand: https://code.google.com/archive/p/mqtt4erl/wikis/QualityOfServiceUseCases.wiki

st0ff3r, 2016-11-29:
I think the queue code should be replaced with a linked list so messages can be stored until successfully delivery (PUBCOMP) and then removed from the queue. The problem occurs if the network silently drops the PUBLISH (QoS == 2), then the whole MQTT messages is never transmitted.

tuanpmt, 2016-11-29:
yes @st0ff3r 
i've implemented it here: https://github.com/tuanpmt/Arduino/blob/master/libraries/ESP8266MQTTClient/src/mqtt_outbox.c
After testing and stability on the Arduino, I will merge it inot esp_mqtt

st0ff3r, 2016-11-29:
> On 29 Nov 2016, at 06.11, Tuan PM <notifications@github.com> wrote:
> 
> yes @st0ff3r
> i've implemented it here: https://github.com/tuanpmt/Arduino/blob/master/libraries/ESP8266MQTTClient/src/mqtt_outbox.c
> After testing and stability on the Arduino, I will merge it inot esp_mqtt
> 
Cool, I am really looking forward to have it in the esp_mqtt version - thanx a lot for the nice work :)

closed 2016-11-30 by tuanpmt