tuanpmt/esp_mqtt #131: MQTT Disconnect - Connect Endless Loop

hannes-angst, 2016-12-18:
I ran into a strange problem, using your code on Sonoff devices:

        (1) Wifi connection, 
        (2) mqqt connect, 
        (3) some mqtt sub (see log below) 
        (4) one pub,  (see log below)
        (5) 1-3 secs 

                   pm open,type:1 0

        (6) mqtt client implementation disconnects 
        GOTO 2

 

Logs from esp8266 (sonoff)

        SDK ver: 2.0.0(656edbf) compiled @ Jul 19 2016 17:58:40
        phy ver: 1055, pp ver: 10.2

        mode : softAP(5e:cf:7f:e8:87:a0)
        add if1
        dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
        bcn 100
        [INFO]

        [INFO] BOOTUP...
        [INFO] SDK version:2.0.0(656edbf) rom 0
        [INFO] Time = 114816
        [INFO] Chip ID: 00E887A0
        [INFO] CPU freq: 80 MHz
        [INFO] Program: user1.bin
        [INFO] Flash size map [2]: 1024 KB (512 KB + 512 KB)
        [INFO] Free heap size: 50432
        [INFO] Memory info:
        data  : 0x3ffe8000 ~ 0x3ffe87b4, len: 1972
        rodata: 0x3ffe87c0 ~ 0x3ffe8db4, len: 1524
        bss   : 0x3ffe8db8 ~ 0x3ffef408, len: 26192
        heap  : 0x3ffef408 ~ 0x3fffc000, len: 52216
        [INFO] -------------------------------------------
        [INFO] Remote Switch v.1.3.0
        [INFO] compile time:Dec 18 2016 15:53:56
        [INFO] -------------------------------------------
        [INFO] CONFIG: Load configuration from flash ...
        [INFO] done
        [INFO] Relay OFF
        [DEBUG] MQTT:InitConnection
        [DEBUG] MQTT:InitClient
        bcn 0
        del if1
        usl
        mode : sta(5c:cf:7f:e8:87:a0)
        add if0
        sleep enable,type: 1
        [INFO] Wifi Status: 1
        scandone
        state: 0 -> 2 (b0)
        state: 2 -> 0 (2)
        reconnect
        scandone
        state: 0 -> 2 (b0)
        state: 2 -> 3 (0)
        state: 3 -> 5 (10)
        add 0
        aid 6
        cnt

        connected with olympus, channel 1
        dhcp client start...
        ip:192.168.13.102,mask:255.255.255.0,gw:192.168.13.1
[... 135 lines cut ...]
        Antwort von 192.168.13.102: Bytes=32 Zeit=4ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=3ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=3ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=1ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=2ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=3ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=8ms TTL=128  <---- mqtt disconnect event
        Antwort von 192.168.13.102: Bytes=32 Zeit=2ms TTL=128
        Antwort von 192.168.13.102: Bytes=32 Zeit=5ms TTL=128

MQTT Output

Every entry is one disconnect -> connect cycle.

        $ mosquitto_sub  -v -t /angst/devices/balkon/#
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}
            /angst/devices/balkon/lampe/info {"name":"balkon/lampe","app":"Remote Switch","version":"1.3.0","mac":"5c:cf:7f:e8:87:a0","ip":"192.168.13.102","type":"switch","status": "off","base":"/angst/devices/balkon/lampe/"}

tuanpmt, 2016-12-18:
This problem with qos=1, can you use with qos=0. I will check and resolve this soon.
Thanks

hannes-angst, 2016-12-19:
Will try this out this evening and posting update.

hannes-angst, 2016-12-19:
Thank you very much. This solved the problem. 👍

closed 2016-12-19 by hannes-angst