tuanpmt/esp_mqtt #134: Ring Buffer overflow causes corrupt data which leads to buffer reset

asadislam94, 2017-01-19:
I was deliberately overflowing the message buffer and I saw that in this scenario, The buffer resets after a certain number of overflows occur. I am queuing 50 messages of total packet length 37. Then I queue 3 packets of length 49,48 and 43 respectively. This is all done while making sure no message is sent during this queuing process.

I have found the cause of the problem.
 ` while (QUEUE_Puts(&client->msgQueue, client->mqtt_state.outbound_message->data, client->mqtt_state.outbound_message->length) == -1) {
    MQTT_INFO("MQTT: Queue full\r\n");
    if (QUEUE_Gets(&client->msgQueue, dataBuffer, &dataLen, MQTT_BUF_SIZE) == -1) {
      MQTT_INFO("MQTT: Serious buffer error\r\n");
      return FALSE;
    }
  }`
When you put something in queue, the underlying code starts saving data without checking if there is enough space. When the buffer fill count reaches maximum, it just returns '-1' without deleting the partially saved data.
This does not cause problem for the initial few overflows,  but when the read pointer of the buffer reaches this corrupt data, it seems to clear the whole buffer. 
I have changed the code to account for this by comparing the message size to available space before trying to call the Queue_PUT function.
`  while ((client->msgQueue.rb.size-client->msgQueue.rb.fill_cnt)<client->mqtt_state.outbound_message->length+2) {
    MQTT_INFO("MQTT: Queue full\r\n");
    if (QUEUE_Gets(&client->msgQueue, dataBuffer, &dataLen, MQTT_BUF_SIZE) == -1) {
      MQTT_INFO("MQTT: Serious buffer error\r\n");
      return FALSE;
    }
  }
  QUEUE_Puts(&client->msgQueue, client->mqtt_state.outbound_message->data, client->mqtt_state.outbound_message->length);`

The reason for adding 2 to the message length is that buffer requires 2 extra spaces to mark start and end.

archenroot, 2017-12-02:
So you fixed that issue by adding 2 to the message?

asadislam94, 2017-12-02:
I think this should be fixed with this pull #126 However I haven't tested that yet.

closed 2023-11-15 by asadislam94