tuanpmt/esp_mqtt #142: dangerous relocation: call0: call target out of range: system_get_flash_size_map

ddze7374, 2017-07-21:
Instead of compiling the source files every time, I  built a static library for the esp_mqtt. A try to link the library with a user_main function stops at a point with a message 

 `
gcc version 4.8.2 (GCC) 
./lib/libesp_mqtt.a(rfinit.o): In function `user_rf_cal_sector_set':
rfinit.c:(.text+0x5): dangerous relocation: call0: call target out of range: system_get_flash_size_map

Do I have to select different bin file to link to or what ? 
`

bclouser, 2017-07-24:
A couple things:
- There is already the means for compiling this repo as a library (though not documented in the readme, i don't blame you for you not knowing about it). You can simply call the esp makefile with target "lib". see this commit 8cf1063d8054268a4720c0d9005fef15a9645a6d 
- Secondly, it would be good to see more of your compile/linker steps. Specifically your final linker command... possibly makefile snippet

ddze7374, 2017-07-24:
I never had a problem in building the static library. Above message occurred  when I tried linking the lib to an application. But the problem is gone (somehow). Thanks

closed 2017-07-24 by ddze7374