tuanpmt/esp_mqtt #148: Problems building for "user/user_main.c" example..

swingwinger, 2017-11-01:
I followed the instructions as best I can tell..  For background, building on a Linux Mint distribution, using ESP8266 NONOS SDK V2.0.0 20160810..
I made a include/user_config.sample.h
I did a make (esp_mqtt/Makefile), and got the following at the bottom of the output:

xtensa-lx106-elf-gcc -L/home/matt/Espressif/ESP8266_SDK/lib -T/home/matt/Espressif/ESP8266_SDK/ld/eagle.app.v6.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -g -O2 -Wl,--start-group -lc -lgcc -lphy -lpp -lnet80211 -lwpa -lmain -llwip -lcrypto -lssl -ljson -ldriver build/esp_mqtt.a -Wl,--end-group -o build/esp_mqtt.out
/home/matt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -ldriver
collect2: error: ld returned 1 exit status
make: *** [build/esp_mqtt.out] Error 1

Since I saw the line below had "driver" in it, and "build/driver/uart.o" appeared to have been made:
SDK_LIBS = c gcc phy pp net80211 wpa main lwip crypto ssl json driver

...So, I removed "driver" from the line above and executed make again to see what would happen...  And the following came out:

touch user/user_main.c
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/home/matt/Espressif/ESP8266_SDK/include -I/home/matt/Espressif/ESP8266_SDK/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_12:49:18_CDT\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c user/user_main.c -o build/user/user_main.o
xtensa-lx106-elf-ar cru build/esp_mqtt.a build/user/rfinit.o build/user/user_main.o build/driver/uart.o build/mqtt/mqtt.o build/mqtt/mqtt_msg.o build/mqtt/proto.o build/mqtt/queue.o build/mqtt/ringbuf.o build/mqtt/utils.o build/modules/wifi.o
xtensa-lx106-elf-gcc -L/home/matt/Espressif/ESP8266_SDK/lib -T/home/matt/Espressif/ESP8266_SDK/ld/eagle.app.v6.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -g -O2 -Wl,--start-group -lc -lgcc -lphy -lpp -lnet80211 -lwpa -lmain -llwip -lcrypto -lssl -ljson build/esp_mqtt.a -Wl,--end-group -o build/esp_mqtt.out
build/esp_mqtt.a(mqtt.o): In function `deliver_publish':
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:120: undefined reference to `espconn_secure_send'
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:121: undefined reference to `espconn_send'
build/esp_mqtt.a(mqtt.o): In function `mqtt_send_keepalive':
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:132: undefined reference to `espconn_secure_send'
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:136: undefined reference to `espconn_send'
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:149: undefined reference to `espconn_abort'
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:153: undefined reference to `espconn_abort'
build/esp_mqtt.a(mqtt.o): In function `mqtt_timer':
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:492: undefined reference to `espconn_secure_send'
build/esp_mqtt.a(mqtt.o): In function `mqtt_tcpclient_discon_cb':
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:500: undefined reference to `espconn_send'
build/esp_mqtt.a(mqtt.o): In function `MQTT_Ping':
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:677: undefined reference to `espconn_secure_send'
/home/matt/Espressif/myProjects/esp_mqtt/mqtt/mqtt.c:679: undefined reference to `espconn_send'
collect2: error: ld returned 1 exit status
make: *** [build/esp_mqtt.out] Error 1

I may have made an error, but haven't been able to figure out what may be missing, since I believe I'm following the build instructions properly.  I had included the library in another project and had the same results, so I decided to isolate the library example to "divide and conquer", and the results were identical.

Thanks.  Looks like a great project and hope I can use it. :-)

harryd100, 2017-11-01:
I'm also using Mint using NONOS SDK 2.0.0 and when executing make i do not get the errors you see. You have to modify the user_config.local.h not the user_config.sample.h
My output looks like:
make
touch user/user_main.c
mkdir -p build/user
mkdir -p build/driver
mkdir -p build/mqtt
mkdir -p build/modules
mkdir -p firmware
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c user/rfinit.c -o build/user/rfinit.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c user/user_main.c -o build/user/user_main.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c driver/uart.c -o build/driver/uart.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/mqtt.c -o build/mqtt/mqtt.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/ringbuf.c -o build/mqtt/ringbuf.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/mqtt_msg.c -o build/mqtt/mqtt_msg.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/queue.c -o build/mqtt/queue.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/proto.c -o build/mqtt/proto.o
xtensa-lx106-elf-gcc -Iuser -Idriver -Imqtt -Imodules -Iuser/include -Idriver/include -Imqtt/include -Imodules/include -Iinclude -I/opt/esp-open-sdk/sdk/include -I/opt/esp-open-sdk/sdk/include/json -g -Wpointer-arith -Wundef -Wl,-EL -Wno-implicit-function-declaration -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -DBUID_TIME=\"2017-Nov-01_20:15:58_CET\" -DMQTT_DEBUG_ON -DDEBUG_ON -D_STDINT_H -DLOCAL_CONFIG_AVAILABLE  -c mqtt/utils.c -o build/mqtt/utils.o
[... cut ...]
xtensa-lx106-elf-ar cru build/esp_mqtt.a build/user/rfinit.o build/user/user_main.o build/driver/uart.o build/mqtt/mqtt.o build/mqtt/ringbuf.o build/mqtt/mqtt_msg.o build/mqtt/queue.o build/mqtt/proto.o build/mqtt/utils.o build/modules/wifi.o
xtensa-lx106-elf-gcc -L/opt/esp-open-sdk/sdk/lib -T/opt/esp-open-sdk/sdk/ld/eagle.app.v6.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -g -O2 -Wl,--start-group -lc -lgcc -lphy -lpp -lnet80211 -lwpa -lmain -llwip -lcrypto -lssl -ljson -ldriver build/esp_mqtt.a -Wl,--end-group -o build/esp_mqtt.out
/opt/esp-open-sdk/esptool/esptool.py elf2image  build/esp_mqtt.out -o firmware/esp_mqtt
esptool.py v1.2

swingwinger, 2017-11-01:
That's my bad in my description... I did make a  "/user_config.local.h".  

So that's *not* the issue.  Errors I have still stand.  Sorry about my error.  I updated my comment above to get things on track. Not having user_config.local.h would have caused other dependency issues...  

Thanks for making the effort to reply..

harryd100, 2017-11-01:
Made a mistake too, i'm using SDK 2.1.0, corrected it above

swingwinger, 2017-11-01:
Thanks again... ...I may try 2.1.0 then...  That may help.  Take me a while to try it and report back.  In the meantime, maybe someone can still see something or has the same issue..