tuanpmt/esp_mqtt #160: Why use of while in MQTT_* functions?

davydnorris, 2019-04-04:
Hi @tuanpmt,

Firstly thank you so much for your library - it's fantastic! 

See for example MQTT_Subscribe:

https://github.com/tuanpmt/esp_mqtt/blob/37cab7cd8a42d51dc9ca448a6eef447ce8ce5b3e/mqtt/mqtt.c#L588

in this function while () is used in line 601 but it appears that it should only ever execute once? This is also the case for MQTT_Subscribe (line 620), MQTT_UnSubscribe (line 648), MQTT_Ping (line 673). Should this be an if() instead?

Also the dataBuffer array will only ever come into play when the buffer is full and there's an error condition, so why allocate the whole array on the stack at invocation? Isn't it better to use os_zalloc and allocate it only if needed?

st0ff3r, 2019-04-07:
the while () { ... if (QUEUE_Gets(...) } is there to remove from the tail of the queue until the new message fits

davydnorris, 2019-04-07:
Ah thank you! Makes sense now, and I also now need to go back and tweak my mods for allocating the buffer so that it doesn't thrash the memory.

closed 2019-05-02 by davydnorris