tuanpmt/esp_mqtt #161: buffer corruption because of unterminated data

davydnorris, 2019-04-06:
https://github.com/tuanpmt/esp_mqtt/blob/37cab7cd8a42d51dc9ca448a6eef447ce8ce5b3e/mqtt/mqtt.c#L304

This line copies the character data that is received from the MQTT broker but leaves it unterminated, leading to buffer corruption later. Suggest adding the line 

`client->mqtt_state.in_buffer[len] = '\0';`

after it, which prevents overruns when using the data directly from the buffer.

someburner, 2019-04-08:
I'm not sure this is suitable to have in the library itself. There is no requirement that you must be passing strings over MQTT.

davydnorris, 2019-04-13:
Exactly. If you're sending string data then the zero termination means you save a copy from the buffer. If you're not, you would be using len to work out how much data you have anyway.

closed 2019-05-04 by davydnorris